lkaino/Triflight #4: GPS settings are unavailable in 0.4 version of triflight

Ultratkinfo, 2016-01-09:
When configuring gps, the following options are unavailable in CLI: gps_provider, gps_auto_config and so on. Also, in cleanflight, GPS options (in Configuration tab) do not persist.

Bengt-M, 2016-02-27:
The GPS fetaure was removed in 476597ca5dc16927e0a3519b0518019d2a0d7dd9 when GTUNE was added .

Based on [this post](http://rcexplorer.se/forums/topic/triflight-setup-guide/page/8/#post-25884) I'd guess the trade-off should be different today. Or maybe that whas a confusion with autotune.

Pls try to enable again.
Before checking other gps issues, obviously :)

If something has to go, to save space, we may need forum opinions for the best compromise.

wolfpackt99, 2016-03-01:
also trying to save gps. maybe i need new firmware. settings in cli are gone. running .4

lkaino, 2016-03-12:
0.5 beta 2 should have working gps

closed 2016-03-12 by lkaino

wolfpackt99, 2016-03-13:
Great news. Thanks.