lkaino/Triflight #7: Tricopter looses control while doing “air jumps”

pqueiros, 2016-01-17:
Tricopter looses control while doing “air jumps”. 

Use case:
- acro mode
- hover the tricopter levelled 
- punch the throttle (100%) for 3-4

Expected result:
- craft camps vertically at “max speed”

Experienced result:
- craft starts vertical climb
- after 1~2 seconds tricopter rolls and yaws easily and looses control.
- pilot could not recover flight and tricpoter crashed.

This occurred two times today, last crash was a really bad one (time to upgrade hw :)). I could not recover the flight in both occasions. It felt like something in the controller saturated somehow.

Normal flight felt quite ok.

Occurrence rate 2/10. Seems that requires a a significant amount of time at full throttle.

 

Source code used was  f6d8753 from master.
cli settings:
[99_tmp_cli_settings_before_autotune.txt](https://github.com/lkaino/Triflight/files/93541/99_tmp_cli_settings_before_autotune.txt)

lkaino, 2016-01-17:
I haven't noticed such behavior, do you have a video or log of the behavior?

pqueiros, 2016-01-17:
unfortunately no.

lkaino, 2016-01-17:
Have you calibrated the ESCs?
17.1.2016 9.00 ip.

pqueiros, 2016-01-17:
ESCs were calibrated when I upgraded to version 0.4 of triglight. I think
Esc issues would show every time I would go full throttle. Which was not
the case. I was also thinking about maybe a bad motor but I would imagine
to notice something in normal  flight.

Since I don't have flight data I suggest to just keep an eye and test this
really well with other setups mine is kind out of order now 😉

lkaino, 2016-01-17:
I can test on mine next time I get to air. Weather is challenging.

pqueiros, 2016-01-18:
is the acc data used while in acro mode? I might have forgotten to calibrate the accelerators after I flashed master.

lkaino, 2016-01-18:
ACC is not used in acro mode.

pqueiros, 2016-02-13:
Closing this issue since could not reproduce it in the rebuilt tricopter, 9 batteries done without issues.

In the post crash analysis I have noticed that one of the front motors was missing the lock washer that presents the shaft from sliding out. I was not sure if this was from before or after the crash. If it was from before it could have been the root cause of the crash since by doing full throttle could trigger the rotor to slide and lock half way making the copter tilt and crash.

I think this was what happened. :).

closed 2016-02-13 by pqueiros

papadkostas, 2016-02-27:
Did you use ground wire for ESC signals when you had this issue?