lkaino/Triflight #8: Gtune feature in triflight

pqueiros, 2016-01-17:
I have tried the Gtune feature as soon as I took off I punched the throttle a little and experienced the issue reported in https://github.com/lkaino/Triflight/issues/7.

Question, is GTUNE supported to work with the new tricopter mixer?

lkaino, 2016-01-17:
I have never tried GTUNE. If there's a problem, it is related to tricopters in general, shouldn't be caused by the tricopter mixer. Does GTUNE even touch yaw axis?

pqueiros, 2016-01-17:
Accordingly with the documentation in clean flights GTUNE should ignore YAW on tricopters

Bengt-M, 2016-01-17:
I have been browsing the src a little.For what it's worth, I'd say GTUNE is enabled. As I read it's purpose is to tune P, not I or D. It looks like it disables tuning for yaw if there are less than 4 motors. It still computes an error but it looks like it doesn't change the P. (Maybe not the best criteria but anyway; yaw tuning should be disabled for a normal tri.)

Intuition says that 'saturated' (as you say  in #7) sounds like I. Have you calibrated acc properly in the FC? I'd guess something like this could happen if you use tx trim instead. Just guessing now...

pqueiros, 2016-01-18:
Hi,
Thanks for the code checkout. From a code perspective looks like that GTUNE would be compatible with the tricopter mixer which is good news :). I guess I will give it another try later on.

closed 2016-04-01 by lkaino