lkaino/Triflight #12: Servo stucked after connecting battery

Nullstr1ng, 2016-02-06:
I am using the 0.5 version of Triflight on my CC3D. By the way, I checkedout the version 0.5 tag and used this command make TARGET=CC3D

So after flashing, cycled the power and connected the battery, the servo was stuck, tilted left (if the tricopter is facing at you) I don't understand why but in betaflight, it's working perfectly.

Can someone take a look at this?

Servo is connected at M4 by the way so I assume this is correct.

lkaino, 2016-02-09:
Unfortunately I don't have a CC3D target to test on. I have already a branch where I have merged to latest Cleanflight. Need to do a bit of testing on that one before releasing a hex for testing. That might fix the CC3D issues also.

If you feel adventurous you could try to build a hex from that branch and test out carefully.

Nullstr1ng, 2016-02-09:
Thanks! I would really like to try TriFlight. Thanks for having a time developing a firmware for our tricopters

Nullstr1ng, 2016-02-15:
servo_pwm_rate didnt help either..

lkaino, 2016-03-12:
Could you test on beta 2?

closed 2017-06-20 by lkaino