lkaino/Triflight #15: Sparky Target in RC5 broken

Ga1j1n, 2016-03-18:
The Sparky hex corrupts the board prompting an unrecognised target message in windows devices, the bootloader bins must be shorted to allow re-flashing of cleanflight.

(It's a shame as I was looking forward to trying it)

lkaino, 2016-03-29:
Too bad I don't have a sparky board to test on :(.

Ga1j1n, 2016-03-29:
I'm happy to test, or could make a donation towards one, I have to admit I'm gravitating towards the Sparky2 and raceflight these days for Quads but my Trifecta Tri is built with a Sparky 1.

http://www.ebay.co.uk/itm/CC3D-Upgrade-FC-Sparky-Flight-Controller-V2-Tau-Labs-10DOF-with-Protect-Case-/271917877268?hash=item3f4f918814:g:2NkAAOSwu4BVk2Kk

http://www.xt-xinte.com/F16821.html

lkaino, 2016-03-29:
Could try to make a build without the servo feedback pins configured. It's
most likely that or then sparky has less flash/ram. Could also make another
build without GPS for testing.

Ga1j1n, 2016-03-29:
Sounds good, let me know and I'll flash it.

Thanks

closed 2017-06-20 by lkaino