lkaino/Triflight #16: Tricopter oscillates in flight when given yaw right input

Ultratkinfo, 2016-03-29:
I am running a somewhat standard setup on David's mini-tricopter:
Triflight-0.5_beta2
Savox SH0255 servo running at 6.0 volt, with a feedback wire connected via voltage divider.
Motors: BE2208-1300 
ESC: 20A running BL-HELI
FC: Naze32 rev6
Props: HQ6x4.5

I did run tailtune, and this is what I got:
set tri_servo_angle_at_max = 400
set tri_tail_motor_thrustfactor = 92
tri_tail_servo_speed = 366

PIDs are out of the box. 
Here is the video of one of my flights: https://www.youtube.com/watch?v=OjOqRi3jyDQ

Drone is very stable and flies great, except for when I give it yaw right. Yaw left works great.

lkaino, 2016-03-29:
Hmm,  hard to say what's wrong. You wouldn't happen to have a blackbox? 

The shaking is so fast that it is most likely caused by motor throttling up and down. The feedback might have some issues when servo is tilting left. Could you try with virtual servo. Disconnect the feedback cable also. 

If that doesn't help try lowering the D and P gains on first roll and then pitch.

Ultratkinfo, 2016-03-31:
Ok, it was the feedback.
When I disconnected feedback wire, servo settings were way off in the left direction. So far off that I couldn't calibrate it without moving some splines.

After doing a tailtune it flies great.

lkaino, 2016-04-01:
So you are not able use feedback on that servo? Maybe this case can be closed.

closed 2016-05-19 by lkaino