lkaino/Triflight #17: Strong tail oscillation

platom, 2016-03-31:
Tried the new 0.5 beta2 fw today on my 190-size tricopter (custom frame).

The yaw oscillated like crazy, changing PIDs doesn't seem to make a difference. Flashed back to 0.4 and everything works perfectly. Same PIDs and controller were used. I am using the standard BMS-210 servo. Awesome work on triflight btw.

lkaino, 2016-03-31:
Whay PID controller did you use? I didn't tune the luxfloat at all. The gains are most likely off. If that's the case, you could try the rewrite controller.

platom, 2016-03-31:
I first tried the rewrite controller, with a fresh installation of 0.5beta2 and I didn't even touch settings as PIDs or looptime etc.

I thought stock PIDs must be optimized for bigger tricopters so I lowered them a lot (p around 2), but the tail oscillaton was still there. 

The problem remained the same even after switching to luxfloat and roughly tweaking PIDs.

I did notice that the oscillation is barely there on descend, it seems to be linear to throttle input. It gets absolutely crazy on punchouts.

Thank you.

lkaino, 2016-03-31:
It might be related to a bug in virtual servo calculation in beta 2, will be fixed in next one.

platom, 2016-03-31:
I tried beta1 too and the problem was still present. Might be something with CF 1.12 as it worked with 0.4 (betaflight)

lkaino, 2016-04-01:
That is very strange as beta 1 contained very little changes to flight behavior compared to 0.4. Beta 1 is still based on same betaflight version as 0.4.

Beta 2 is based on CF 1.12.

Do you have a video or blackbox log of the scenario?

platom, 2016-04-01:
Oh you are right then, I probably never tested beta 1, let me test it later today, will report results.

platom, 2016-04-19:
Managed to destroy me servo. Waiting for new gears. Sorry.

kristoferahlqvist, 2016-05-17:
I have the same issue. Crazy tail oscillations with 0.5b2 no matter the pids, but with 0.4 it flies fine

Bengt-M, 2016-05-17:
The code correction was done in commit 33dbe9c634f3980e7896a118d97df9dc5212b1c2 in the master branch. Your options are:
- add the hardware servo feedback and use 0.5-Beta2
- use 0.5-Beta1
- if you are brave, make your own unofficial hex (or use an unofficial hex from someone else)
- wait for the next official release

lkaino, 2016-05-17:
I'll try to get the next beta out this week, it should fix these issues.

platom, 2016-05-25:
Just noticed beta3 is out, awesome!
Will try it out and report back asap. Great job!

bremsstrahlung, 2017-04-18:
I noticed this behavior also on 0.7b1.
The issue seems to be that the parameter tri_tail_servo_speed is used even when tri_servo_feedback=VIRTUAL. If I set tri_tail_servo_speed=0 (default 300) The issue disappears.

lkaino, 2017-04-18:
The servo speed is only used in virtual feedback mode. You don't have the feedback signal connected?

I didn't test the virtual feedback mode. Is the servo oscillating all the time?

bremsstrahlung, 2017-04-18:
My servo don't have a feedback cable and I haven't bothered to add one yet so I'm currently running with tri_servo_feedback=VIRTUAL. With tri_tail_servo_speed=300 the servo goes between what seems to be my tuned min and max as soon as the motors spin up, even before I apply enough throttle to actually lift off.

ResinSmoker, 2017-11-16:
After updating, I've been seeing oscillations both at idle and forward flight.
Made a video of the test flight: https://www.youtube.com/watch?v=PM1rwsYw8G4&t=144s

Note: Testing yaw / roll mixing but wag can be seen clearly with mixing active and disabled.

```
# Triflight 0.7 Beta 3 / RCEXPLORERF3 3.1.7 Oct 21 2017 / 13:57:41 (17fa321)
servo 5 1144 1998 1577 40 40 100 -1

serial 0 0 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 2 1 115200 57600 0 115200
aux 0 0 0 1800 2100
aux 1 1 2 1700 2100
aux 2 2 2 1300 1700
aux 3 31 1 1800 2100
set min_check = 1010
set max_check = 1990
set min_throttle = 1100
set max_throttle = 1900
set align_board_pitch = 180
set gyro_sync_denom = 2
set servo_lowpass_hz = 300
set servo_lowpass = ON
set tri_tail_motor_thrustfactor = 60
set tri_tail_servo_speed = 216
set failsafe_off_delay = 20
set failsafe_throttle = 1150
set failsafe_procedure = AUTO-LAND
set acc_hardware = MPU6000
set acc_lpf_hz = 4
set acc_trim_roll = -6
set baro_hardware = MS5611
profile 0
set yaw_lowpass = 500
set p_level = 25
set i_level = 25
rateprofile 0
rateprofile 0
set rc_rate = 50
set rc_rate_yaw = 50
```