lkaino/Triflight #25: Tuning questions

Bengt-M, 2016-04-27:
I am trying (maybe too much) to understand this tuning.
The code is complex and it's not so easy to translate back into normal math. So I made a little sketch.
[Geometry.pdf](https://github.com/lkaino/Triflight/files/238447/Geometry.pdf)

Servo angle A from vertical
X = Thrust force. Horisontal component = X * sin(A)
Y = Torque force (perpendicular to X). Horisontal component = Y * cos(A)
These must be in balance, so Y * cos(A) = X * sin(A)
Using X/Y = K, leads to K = 1/tan(A)

Example numbers:
K = 13.8  => A = 4.1 deg
Hm..

I have ignored pitch. I see that the lifting force will vary by A to require a pitch compensation, but I don't see how that would affect yaw.

If this is correct the tuning calculation can be simplified.

I feel I'm missing something important here so I don't know if I should trust my own words. Please tell me what is wrong.

lkaino, 2016-04-28:
Well dang, you might be correct that the pitch compensation need not to be taken into account in the tuning. The compensation does affect yaw output, but shouldn't affect the thrust factor. The factor should remain more or less constant. 

I need to double check still when at computer but your equations seem correct. 

You could verify the other math at some point also.

Bengt-M, 2016-04-29:
You see where this is going, right?
To start with, the complex iteration you now have at state TT_WAIT_FOR_DISARM could be replaced with one simple 1/tan.
But then the next idea comes creeping up; it should be just as good to turn the tuning approach around. Now you assume the mid is good and tune the factor, but it should work just as fine the other way too.
Someone with a fancy test stand could measure the factor once and for all, since it should be a property of the prop type and independent of copter build. Then the tuning could aim at finding the right mid.
This can be automated but with the laptop at the field this can be done now without any code change; just turn off the I term with a profile, use trims to stop the spinning, adjust mid based on trim, reset trim and repeat until good.
I will try this when weather permits.

A first step of sw support could be a small redesign of the bench tuning so the mid is changed just one tick at a time, with a confirmation beep, just like the acc tuning does. With that the tuning above can be done without laptop.

There is no prestige here so prove me wrong and I'll learn something. If this idea is wrong then there just have to be test cases where the mid tuning and factor tuning can be verified separately in flight test. I started to define those tests but I have not managed yet, so now I'm leaning to the position that it's only the sum of static errors that matters.

Also; The wind from the prop will hit the copter tail. That may not be perfectly symmetrical and will thus introduce a yaw force. First approximation is to accept that the best servo mid is not exactly vertical. This might mean that the thrust and the torque forces are not exactly perpendicular, but I would think it's good enough approximation since, for small angles, tan(a)=a anyway.

(And thinking about speed I also think of the [Dissymmetry of lift](https://en.wikipedia.org/wiki/Dissymmetry_of_lift). Maybe the resulting roll force can be modelled, but first things first) 

We can't have a model full of unknowns and we can't control what we can't measure. At some point we just have to let the I term do the rest but it would feel sound if the I-term was 0 (i.e. the sum effect from all static errors was 0) in the most typical flight scenario.

closed 2017-06-20 by lkaino