lkaino/Triflight #30: Merge changes back into Cleanflight

hydra, 2016-05-12:
Hi guys!

Looks like you guys are doing some good work here.  Cleanflight v1.13.0-RC1 was created yesterday and is a stable codebase from which to further develop your changes.

Could you start creating pull requests so that we can get the TriFlight changes merged back into Cleanflight if you think they are ready?

lkaino, 2016-05-12:
Hey Dominic and thanks for showing interest!

I read your comment about 1.13 and will definitely do a rebase soon, need to familiarize with the new configuration architecture.

Some of the Triflight features are proven to work and for those I could start creating pull-requests. But we only just started on a unit test project and the documentation is non-existent so there's still work to do before they are good enough for CF.

The next version will contain some bigger changes, but I wish after that we can stabilize the features and get them back to CF.

Bengt-M, 2016-05-12:
Totally agree and would be glad to assist.

Maybe a two-step approach could be in place?
To first do one merge to pave the road. Fixing all dependencies and side things but without actually changing the function to the Triflight magic. Can remove a lot of small merge conflicts and make next re-basing and merge more easy.

oxivanisher, 2016-05-27:
I would so love you guys if you can merge the forks together! Opensource development as its finest! Keep up the great work @lkaino and @hydra. :+1:

cmanley, 2016-06-28:
Please include the improved GPS features of the inav port too :)

hydra, 2016-06-29:
@lkaino merging is quite trival.  put your new settings (if any) in one or more structures, add a PG ID (into parameter_group_id.h)  use PG_DECLARE in a header file, for the structures.  use PG_REGISTER in a C file.  Then change any usages from `master_config.somestruct.value` or `profile_master.somestruct.value` to `someStruct().value`.

Look at existing code for working examples.

oxivanisher, 2016-08-02:
Is there a status update on this? Really looking forward to use this with cleanflight again.

kangsterizer, 2016-09-13:
I think the author will need to make PRs of separate pieces of code. I just did a local merge to inav and theres a good bit of hard coded values specifically for the author's copter and boards.

Other than than its not all that much work though I'm pretty much unaware of which features should be considered stable/mergeable or not ;-) (I'm sure the author knows this like 1000x better than me, so just hoping he reappears ;P)

shoreflyer, 2017-04-11:
@lkaino what is the status on merging this back into cleanflight? I've seen mention in the README rebasing onto iNav and/or betaflight?

lkaino, 2017-04-14:
Triflight is now based on betaflight (cleanflight also). I cleaned out some dirty functionality on the way. It's not far from being ready to merge. Would like to get following done:
- get the unit tests updated and some new ones written.
- finalize the new features in 0.7.
- update the documentation
- would like to get rid of the restriction of tail motor spin direction (not absolutely necessary for the merge)