lkaino/Triflight #36: Cant control servo with 0.5b3

kristoferahlqvist, 2016-05-23:
So I tried the new 0.5B3 today but I cant control the servo. It's all the way to the right and nothing happens when I use yaw. I cant tune the yaw with yawtuning either.

I use naze32 and a servo without feedback

kristoferahlqvist, 2016-05-23:
Is there some change how yawtuning works while disarmed?

lkaino, 2016-05-23:
Do you mean the servo setup in unarmed tail tune mode? It should work as before. I didn't test without feedback. Could you post your CLI dump?

kristoferahlqvist, 2016-05-23:
Yeah unarmed tail tune mode. The servo does not move in either unarmed tail tune mode or when I move yaw stick when unarmed.

I have flashed back to 0.5b1 so that I could fly again. I will flash 0.5b3 again and post the dump later

Bengt-M, 2016-05-23:
This is probably the same issue that was fixed in af7f2c76edfb7920fbefaffd2bd016c039797025, which was after 0.5b3.
If you can build your own hex, the current master branch is better.
Otherwise you have to do that tuning the old fashion way as described well by @LeoTheHuman:

Go to the servo tab of the configurator and enable “live mode”. Servo 5 is the tail servo for the tricopter, so we will only be working with its values. To test the deflection at the given value, set all three parameters (mid, min and max) to that value.  ... I would recommend starting with the mid and min values. To find max start with the value of (mid * 2 - min), which should be spot on, but might need minor corrections to it. When you find the values for all three positions, put these into the corresponding fields of Servo 5 and click “Save”.

kristoferahlqvist, 2016-05-24:
I tried to change the values in the servo tab but I dont know how to set it up when i cant get the servo to move while disarmed.

Another weird thing with 0.5b3 is that I cant arm in angle or horizon. Only in acro mode.

Here is the dump:

# version
# TriFlight 0.5 Beta 3/NAZE 1.12.0 May 22 2016 / 19:55:48 (536747b)
# dump master

# mixer
mixer TRI
mmix reset
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature RX_PPM
feature MOTOR_STOP
feature FAILSAFE
feature ONESHOT125

# map
map AETR1234

# serial
serial 0 1 115200 57600 0 115200
serial 1 0 115200 57600 0 115200

# led
led 0 15,15:ES:IA:0
led 1 15,8:E:WF:0
led 2 15,7:E:WF:0
led 3 15,0:NE:IA:0
led 4 8,0:N:F:0
led 5 7,0:N:F:0
led 6 0,0:NW:IA:0
led 7 0,7:W:WF:0
led 8 0,8:W:WF:0
led 9 0,15:SW:IA:0
led 10 7,15:S:WF:0
led 11 8,15:S:WF:0
led 12 7,7:U:WF:0
led 13 8,7:U:WF:0
led 14 7,8:D:WF:0
led 15 8,8:D:WF:0
led 16 8,9::R:3
led 17 9,10::R:3
led 18 10,11::R:3
[... 277 lines cut ...]
set gtune_loP_yw = 10
set gtune_hiP_rll = 100
set gtune_hiP_ptch = 100
set gtune_hiP_yw = 100
set gtune_pwr = 0
set gtune_settle_time = 450
set gtune_average_cycles = 16

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 100
set rc_expo = 85
set rc_yaw_expo = 83
set thr_mid = 50
set thr_expo = 0
set roll_rate = 55
set pitch_rate = 55
set yaw_rate = 55
set tpa_rate = 45
set tpa_breakpoint = 1500
set tri_dynamic_yaw_minthrottle = 250
set tri_dynamic_yaw_maxthrottle = 60

Bengt-M, 2016-05-24:
I see now that my previous response was wrong. Sorry. You have the latest version.

When I compare your settings with my own I see some differences:

your channel map is map AETR1234
Mine is TAER1234
Check so you use a map matching your radio.

servo_pwm_rate
You have 50, I have 250
Guess it depends on servo.

align_board_pitch and align_board_yaw
You have (0, 90), I have (180, 0)
If you have an RCE copter with integrated NAZE you should change but maybe you have something custom built.

Your servo settings are different
You have servo 5 1500 2000 1500 45 45 100 -1
Same value for min and mid can't be right.

You use smix reverse.
If you have a standard RCE kit you should not.
There have been issues reported about servo reversing so it's possible there is still a bug here. I would try reversing in the Tx instead but I'm not at all sure about that. Try a normal servo maybe? 

And also: One big difference between acro and stabilised is that stabilised use the accelerometers. Try redo that calibration and check that the copter is about level when you start.

Hope something helps you.

kristoferahlqvist, 2016-05-24:
To start with. Everything works fine with 0.5b1.

I use aetr1234 cause that's how my radio is setup and everything is correct in the receiver tab 

I have 50 hz pwm rate cause I use an analog servo

I have a 3d printed tricopter and the fc is rotated so I can reach the USB so that's why I have 90 degree yaw offset

I have not been able to setup the servo settings cause I can't get it to move with 0.5b3.

I use six reverse to make the servo move correct direction

I'm aware of the difference between stab and acro.  I have maybe twenty race multirotors before this tricopter. 

The problem is the 0.5b3 firmware.

Issue 1: servo don't move when unarmed
Issue 2: can't arm with stabilised flight modes

Bengt-M, 2016-05-24:
You have no idea how many people start by ignoring half of the instructions and then complain about the performance. You seem to know better. Good for you. Then I'm sure you also know that faultfinding cannot be successful in a system where the fault cannot be seen.
My own tri arms and fly well in stabilised modes and bench tuning works as advertised. I use 0.5b3 so logically there must be some critical difference between my tri and your tri. HW or setup.
It's impossible to guess; I tried to detail the differences between your dump and my dump so you could get some ideas.
A likely alternative on my list is the reversing bug (which I don't think have been squashed for sure yet). Trying a different servo which doesn't need reversing has worked for other people. Actually, removing the reversing setting for a bench test might be helpful. It won't fly that way but maybe it can tell if this bug is still present.
One big difference between 0.5b1 and 0.5b3 is the underlying Cleanflight version. If you like you can try with Cleanflight v1.12.0 and then V1.13. That could also give more info.
No more ideas for now..

lkaino, 2016-05-24:
@Bengt-M gave some good advise. 

It might also be a problem with the virtual feedback as I haven't tested it. Are you Bengt also running a feedback wire?

kristoferahlqvist, 2016-05-24:
First I'd like to thank you for your help Bengt. Seems like you were onto something with the servo reversing. Without servo reversing everything seems to works except that I can't move servo when unarmed. 

But I can move it when tailtuning is activated so that's not really an issue. 

The only issue now is that the servo goes wrong way

Bengt-M, 2016-05-24:
I am not using the servo feedback wire.
0.5b2 had a bug but with 0.5b3 it works fine for me. Tuning and all.

My biggest problem now is that a tree close to my field very aggressively attacked my tri and brutally ripped off the logging connection. So now I have no USB either. But it still flies.

kristoferahlqvist, 2016-05-24:
So how do I reverse servo with 0.5b3?

lkaino, 2016-05-24:
Unfortunately you can't, see #11.

kristoferahlqvist, 2016-05-24:
Bummer :(

lkaino, 2016-05-27:
I'll close as duplicate.

closed 2016-05-27 by lkaino

lkaino, 2016-06-03:
@kristoferahlqvist now it is possible to reverse the servo in the new alpha release

kristoferahlqvist, 2016-06-03:
Great! Thank you!