lkaino/Triflight #37: Use servo configuration angles

lkaino, 2016-05-24:
Instead of specific CLI parameter, use the servo 5 configured max angles (-45, 45 by default). Change the defaults to -40,40.

These parameters were not available when the CLI parameter was implemented.

Bengt-M, 2016-05-24:
I've been thinking the same way so couldn't resist to make a PR
See if you like it.
The servo configured values are not (-45, 45). They are positive in both directions. I'm not sure this is correct. Vaguely remember some issue in the first manual version of tailtune where the configurator seemed to have a different idea.
In the PR I still depend on symmetry in the same way as before. If that is still OK the negative is no issue.

lkaino, 2016-05-30:
Fixed in #38

closed 2016-05-30 by lkaino