lkaino/Triflight #42: tri_motor_acc_yaw_correction: motor cut in fast forward flight

DeepnineRC, 2016-06-29:
Hi,
after reading the release note for 0.5 Beta 3 i thought about the problem cutting the motor in fast forward flight.
Some ideas came to my mind:

1.) Disabling damped light in the ESCs so the motor will not stop that fast. (floaty feeling?)
More time for yaw correction?

2.) Detect fast motor cut and slow down motor after the yaw was corrected. (floaty feeling too?)

3.) The yaw mechanics turn the motor to the left to compensate a CCW rotation. But the motor is not only rotating to the left. It's also moved to the left which gives a force to the tail boom and push the  tail boom to the right ( more CCW turning force). 

Motor off:
https://www.youtube.com/watch?v=NxXuu7b3hxc

Motors on:
https://www.youtube.com/watch?v=VuT6SHdVV3Q

another observation its not only yawing its rolling a little bit. Seen on the second video.

Solution: change the turn point of the yaw mechanism up to compensate the effect or even turn it to the other side and help yawing. Impossible tilt mechanism might be the solution.
This is the next video test i will so.

Another Idea: If you cut throttle not on all motors at the same time. You have two CCW and one CW motors to Yaw a bit. Do a Yaw like on the quadcopter with different rotation speeds. If air is fast and the propellers dont have much grip it might not roll/nick very much.

My flying skills are not good enough to feel the problem in real world flying.  I'm happy not crashing, but i can change a broken prop in under 10 seconds now :-)

I like doing bench tests. If you have another idea for testing. Tell me.

Tell me what you think.
Thanks Jan

lkaino, 2016-06-29:
Hello,

Thanks for showing interest in this tricky issue :).

1) It would help, but damped light really improves the flight in other areas.

2) It would produce a floaty feeling. We could add a limiter of how quickly the motors are allowed to decelerate, but it's more of a hacky workaround. Also quick dives to e.g. hit a gate would be more challenging.

3) This is a known thing with the RCExplorer tilt mechanism, but I don't think it plays a role in this one.

Really nice jig you built there, and nice videos! Did you use the `tri_motor_acc_yaw_correction` there? Try turning it way up, cut the throttle from full on your transmitter and see what axises are affected the most.

Use for example these values:
```
set tri_dynamic_yaw_minthrottle = 250
set tri_motor_acc_yaw_correction = 32
```

DKDarkness, 2016-06-29:
I've had a look at this issue using "the impossible tilt" that more or less takes the mass of the motor out of the equation. Tricopter still has the floaty/tumbling quality. I suspect that the yaw does not have enough authority at low RPM. I have tried to compare the effects of AIRMODE on quadcopter vs. tricopter. When airmode is enabled on a quad you can still maneuver the quad and even do flips and rolls at idle throttle. On the tricopter this is not the case. If you can't control the tricopter at idle or close to idle throttle neither can the flight controller?

lkaino, 2016-06-29:
I have to disagree. Tricopter can do flips and rolls just fine at idle throttle. Nowadays it can even do yaw. Can you clarify what you mean by this is not the case on tricopters?

Bengt-M, 2016-07-01:
3)
Look at the first posting in #25. A yaw effect can only appear if the horisontal components of the forces change due to that sideways offset. I don't see how that can be
(unless we start thinking about turbulence around the servo and the boom but that is not feasible to model anyway).
The offset will give some torque around the roll axis (but probably insignificant compared to other effects when flying).

lkaino, 2017-06-20:
This starts to be more or less solved nowadays. With the latest alpha builds I can cut the throttle without disaster, even with the correction at 0.