lkaino/Triflight #44: No servo output on SPracing f3 evo

Matjordan, 2016-08-03:
I cant get a tail servo to move on any of the 5 non motor outputs. it just goes to center pos and stays there. I think that servo 5 does not get mapped to a PWM output anywhere for that board.

lkaino, 2016-08-03:
It should be the same pin than on Cleanflight. Anyone with that board could assist?

Matjordan, 2016-08-03:
It does not work with the cleanflight either. I did raise an issue over there too

zsoltm, 2016-11-12:
I have similar issue bot on the SPRacing F3 *non-EVO*, and for me it works with CleanFlight 1.14.0.  With "vanilla" CF the mapping is rather odd: motor 1-3 on pin 1-3 and servo on pin 7 (instead of the usual servo on pin 1, motor 1-3 on pins 3-7 that I had on the naze) . Any clues?

bremsstrahlung, 2017-04-18:
In version 0.7b1 the resource mapping seems to be different than earlier. On my Flip32 board I had to remap motors 1-3 and servo 1-2 using the "resource" cli command. This fixed the issue for me

lkaino, 2017-06-04:
Please test with 0.7 Beta 2 release.

bobthemac, 2017-06-20:
I have just flashed 0.7 Beta 2 onto my spracingf3 evo and I am not getting any servo response. Looking at the resource list I think servo one should be on port six although I did try port 5 as well and no response.

```
# resource list
IO
A00: MOTOR 1
A01: MOTOR 2
A02: MOTOR 3
A03: FREE 
A04: ADC_BATT 
A05: ADC_CURR 
A06: SERVO 1
A07: SERVO 2
A08: FREE 
A09: FREE 
A10: FREE 
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: FREE 
B00: SERVO 3
B01: SERVO 4
B02: ADC_EXT 
B03: SPI_SCK 1
B04: SPI_MISO 1
B05: SPI_MOSI 1
B06: I2C_SCL 1
B07: I2C_SDA 1
B08: LED 1
B09: MPU_CS 
B10: FREE 
B11: SERIAL_RX 3
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: SDCARD_DETECT 
C15: BEEPER 
F00: FREE 
F01: FREE 
F04: FREE 

DMA:
DMA1 Channel 1: FREE
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: FREE
DMA1 Channel 5: FREE
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE
DMA2 Channel 1: ADC
DMA2 Channel 2: FREE
DMA2 Channel 3: FREE
DMA2 Channel 4: FREE
DMA2 Channel 5: FREE 

```

lkaino, 2017-06-20:
Please check this page https://github.com/lkaino/Triflight/wiki/Servo-output-channels-for-targets. The servo output for spracingf3 should be 7.

bobthemac, 2017-06-20:
On that page it says its 5-8 for the EVO and not one of them works I am getting no response.

dragonclips, 2017-07-17:
Hello,  i have flashed 0.7 Beta 2 on a SPF3 and the servo connected  in the 7 pwm  of the board (listed as servo 5 in the SERVO TAB as default) it works but moving in the opposite direction, i tried reversing with  -100 rate but seem doesn't have any effect. Any solutions?

dragonclips, 2017-07-17:
Im trying now to edit a normal BETAFLIGHT 3.1.7 for tricopter, just to see what happening

bobthemac, 2017-07-19:
@dragonclips I sugest you open a new issue as this is an unrelated problem.

dragonclips, 2017-08-03:
Hi, i confirm that with the BETAFLIGHT 3.1.7 all working good, i moved the servo to pin 8 istead of 7 and in the servo tab 5 all working correct.