lkaino/Triflight #49: SERVO not working with TriFlight V0.7 beta1 on Naze32

bbc25185, 2017-04-25:
I tried everything but I could not get the tail servo working at all (No signal)
My setup:
Flip32 (same as Naze32)
Receiver connected with PPM (Gui see it working correct)
Motor connected to OUT1, 2, 3 (i checked it work)
but i tried all pin... Servo tail not working
If i check in CLI, the resource list is this:
# resource list
IO
A00: PPM 
A01: FREE 
A02: FREE 
A03: FREE 
A04: ADC_BATT 
A05: FREE 
A06: FREE 
A07: ADC_EXT 
A08: MOTOR 1
A09: SERIAL_TX 1
A10: SERIAL_RX 1
A11: MOTOR 2
A12: BEEPER 
A13: FREE 
A14: FREE 
A15: FREE 
B00: FREE 
B01: FREE 
B02: INVERTER 1
B03: LED 1
B04: LED 2
B05: FREE 
B06: MOTOR 3
B07: FREE 
B08: FREE 
B09: FREE 
B10: I2C_SCL 2
B11: I2C_SDA 2
B12: SPI_CS 2
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C13: MPU_EXTI 
C14: FREE 
C15: FREE 

DMA:
DMA1 Channel 1: ADC
DMA1 Channel 2: FREE
DMA1 Channel 3: FREE
DMA1 Channel 4: SERIAL_TX 1
DMA1 Channel 5: SERIAL_RX 1
DMA1 Channel 6: FREE
DMA1 Channel 7: FREE

I tried to manualy assign a resource to SERVO 6, but nothing... (i tried on some pin)
I tried to lower the servo refresh rate... nothing
its a bug????
PS: I tried TriFlight V0.5... Servo work on OUT2.... but the GUI (Cleanflight and Betaflight) not allow me to configure without CLI
PPS: I tried to remap all the OUT (Motor to OUT 4, 5, 6) and the servo to OUT 2... NO working...

odi-um, 2017-04-25:
i've xracer with 0.7b1 and two servos (yaw+pitch). was able to achieve working setup only in such way
1 - yaw servo
2 - pitch servo
3 - motor 1
4 - motor 2
5 - motor 3

Hope this helps you.

lkaino, 2017-04-25:
This post might be helpful: https://rcexplorer.se/forums/topic/triflight-0-7-beta-1/page/3/#post-37397

bbc25185, 2017-04-25:
Ok... i found the error...
Its Ok to set the servo and motor to:
Motor 1 = OUT 6
Motor 2 = OUT 5
Motor 3 = OUT 4
Servo = OUT2

The error??? In the resource,  the reference is to SERVO 1... NOT SERVO 6...
I write "resource servo 1 a11" and all work...
Dont use servo 6 in the resource!!!

odi-um, 2017-04-25:
@lkaino thanks for pointing out, now its obvious why its working in such way. bad that we came to that by "scientific method".

@bbc25185 yes, forgot to mention that i was tried the same way, resource numbers as 1 for pitch and 6 for yaw (like it should be) but again using "scientific method" found out that should be 1 for yaw and 2 for pitch.

lkaino, 2017-06-04:
Please test with Beta 2, should work out of the box now.

lkaino, 2017-06-20:
Closing, reopen if still an issue.

closed 2017-06-20 by lkaino