lkaino/Triflight #57: Problem with external I2C on Omnibus F4 boards

samp38, 2017-10-28:
Hi,
First, thanks for your work. It's amazing how Triflight drives my Trifecta. However, i can't remap I2C (whose pins are shared with UART3 on the Omnibus F4) to B10 - B11 pins. It seems that i2C is not active on the Omnibus F4 target. Is there a way around? I need to connect an i2C magnetometer to my flight controller.

Thanks

Here is my resource output:

resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A08
resource SERVO 1 A01
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07

resource list:
# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE 
A01: SERVO 1
A02: FREE 
A03: MOTOR 3
A04: MPU_CS 
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE 
A09: FREE 
A10: FREE 
A11: USB 
A12: USB 
A13: FREE 
A14: FREE 
A15: OSD_CS 
B00: MOTOR 1
B01: MOTOR 2
B02: FREE 
B03: BARO_CS 
B04: BEEPER 
B05: LED 1
B06: FREE 
B07: SDCARD_DETECT 
B08: FREE 
B09: FREE 
B10: FREE 
B11: FREE 
B12: SDCARD_CS 
B13: SPI_SCK 2
B14: SPI_MISO 2
B15: SPI_MOSI 2
C00: FREE 
C01: FREE 
C02: ADC_BATT 
C03: FREE 
C04: MPU_EXTI 
C05: FREE 
C06: FREE 
C07: FREE 
C08: FREE 
C09: FREE 
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: FREE 
C14: FREE 
C15: FREE 
D00: FREE 
D01: FREE 
D02: FREE 
D03: FREE 
D04: FREE 
D05: FREE 
D06: FREE 
D07: FREE 
D08: FREE 
D09: FREE 
D10: FREE 
D11: FREE 
D12: FREE 
D13: FREE 
D14: FREE 
D15: FREE 

Currently active DMA:
--------------------
DMA1 Stream 0: FREE
DMA1 Stream 1: FREE
DMA1 Stream 2: FREE
DMA1 Stream 3: FREE
DMA1 Stream 4: SDCARD
DMA1 Stream 5: FREE
DMA1 Stream 6: FREE
DMA1 Stream 7: FREE
DMA2 Stream 0: FREE
DMA2 Stream 1: FREE
DMA2 Stream 2: FREE
DMA2 Stream 3: FREE
DMA2 Stream 4: ADC
DMA2 Stream 5: FREE
DMA2 Stream 6: FREE
DMA2 Stream 7: FREE

samp38, 2017-10-31:
I tried to add i2c definition in target.h but there is no way to get I2C2 working on pins PB10 and PB11. As soon as i add "#define USE_I2C" in target.h the vcp doesn't appears on startup. Does anyone had the same kind of issue? Thanks.

lkaino, 2017-12-04:
I'm working on rebase on BF 3.2.2. Hopefully the I2C will work there.

samp38, 2017-12-06:
Great news!

Thanks for your work and let me know when this is done!

Sam