lkaino/Triflight #59: Safety Issue: Motor #1 surges upon loss of Servo Feedback (Triflight 0.7 Beta 3)

ResinSmoker, 2018-05-01:
Had my Tricopter-Mini Flip upon Arming due to Motor #1 instantly surging to 100% throttle. After recovering the craft and doing in-house testing I soon discovered that the loss of the Servo feedback voltage resulted in such a surge. (0 VDC) I can only surmise that my Servo's quick disconnect adaptor wasn't fully seated and is the likely cause of the Flip.

Question: Would it be possible to script an event upon the loss of the feedback voltage, where the copter either refuses to arm or switches to virtual mode?

lkaino, 2018-05-06:
Hmm, that should have been fixed by https://github.com/lkaino/Triflight/pull/56/commits/3da51cc041504369eac0335b3b408dadf0ee597e.

It checks that the servo feedback is giving a reasonable angle, and disables the yaw boost if not.

ResinSmoker, 2018-05-06:
Far as I could tell, the Servo wasn't moving at all so I think the FC was trying to compensate for the lack of counter Yaw by increasing the Tail motors speed. Motor #1 (Tail motor) being that it was spun up at 100%, took the brunt of the impact from the flip. The impact was enough to cause Motors #1 & #3, to start Desyncing badly during a later test and check flight. https://www.youtube.com/watch?v=Gd6hDRKj080

Note(1): Everything looked fine on the bench before the test flight.
Note(2): No serious damage other then I had to swap out all the F80's for the RS2203's.