lkaino/Triflight #60: Cannot reverse yaw servo

MSmeets94, 2018-05-27:
I've build a tricopter with a Kakute F4 V2 FC. I've build my own triflight hexfile based on the BF 3.2 rebase.
Everything works fine, but the servo is moving in the wrong direction. Changing the rates to -100% in the servo tab of BF doesn't work. Is there any other way to reverse the servo that i missed, or is it not implemented or broken in the new firmware?

Mattbill1, 2019-06-17:
I have the Kakute F4 V2 as well in a RCExplorer Tricopter LR, and I want to install Triflight.  Still looks like they only support the v1 officially, mind sharing your hex file?

MSmeets94, 2019-07-28:
Hi Matt,

If you are still interested i can give you the hexfile. I recompiled it yesterday after i expiremited with BF 4.0.4 but i couldn't get it to work. Use at your own risk ofcourse