lkaino/Triflight #62: Beta 3 Yaw Center Stick Vibrations

hgg2k, 2018-06-22:
Hi,

I have finished building the Quanum Trifiecta with the F3 Deluxe board.
In beta 2 I had a problem with the yaw servo turning left every time I lowered throttle,
but in beta 3 it has been 99% solved!    Great work.

Now I have another problem.  When the servo is centered it starts oscillating badly,
but when I start turning it to the left or the right, the oscillation goes away.  During
flight the problem is still there but to a lesser degree.

Small clip showing the problem : https://youtu.be/ferMhxt82DA

I am using the EMAX ES09md digital servo but I have not used any feedback wire.
Do you think that this might help to solve the problem?  Any other suggestions?

Thank you.

vicdzen, 2018-06-23:
Did you try to tune yaw PIDs? P and D mostly

ResinSmoker, 2018-07-08:
Your I term determines how aggressively the Yaw will attempt to move with the RC commands. The I-Term acts to stabilize the copter and help prevent overshoot to some degree. The D-Term acts as sort of a system dampener, thus smoothing out the whole system at the cost of increased noise.

I'd suggest P60 / I30 / D20 as good starting points for Yaw. Beyond that, try to keep your D low and only increase the I if the copter tends to drift. Keep your filtering either at the defaults or such that you don't introduce a lot of latency into the system. After that do a flight with a lot of stick movements and pull your BF log. Use this to create a Plasmatree plot to best estimate what Terms need to be adjusted further.