lkaino/Triflight #66: Add new board - Crazybee f3fs

gokalpsterr, 2019-01-06:
Crazybee F3FS is a great all in one board and a great opportunity to make the baby tricopter, micro baby tricopter!

There exists a betaflight for it already:
https://github.com/betaflight/betaflight/blob/master/docs/boards/Board%20-%20CrazyBeeF3FS.md

mikeller, 2019-01-07:
@gokalpsterr: Just wondering, how do you connect the tail servo to this board?

gokalpsterr, 2019-01-07:
That is the greatness about this board, there is a fifth pwm output pin on this board allowing to drive another servo or motor or etc. This board is the black one (f3fs) there is also a red version (f3fr) which i have no idea if that one has an extra pwm as well or not.

gokalpsterr, 2019-01-07:
@mikeller: *correction* my apologies, the black one is f3fs pro and f3fs and f3fr are the red one's versions. But I have no clue what their differences are. I want support for the crazybee f3fs pro (black one) as that one has a 5th pwm pin ready to use.

jbman, 2019-05-11:
I have the red f3fs version with PWM output pad. Figured out that the PWM output pad is pin B06.